InternRobotics/InternUtopia #2: GRUtopia/demo/h1_house.py with JSON decode error

qZhang88, 2024-07-20:
run `python ./GRUtopia/demo/h1_house.py`, met following problem.

```
[2024-07-20 07:29:06,506][ERROR] /isaac-sim/GRUtopia/grutopia/core/task/task.py[line:79] -: h1_house_0
2024-07-20 07:29:06 [1,747,310ms] [Error] [root] h1_house_0
[2024-07-20 07:29:06,506][ERROR] /isaac-sim/GRUtopia/grutopia/core/task/task.py[line:80] -: Expecting value: line 1 column 1 (char 0)
2024-07-20 07:29:06 [1,747,310ms] [Error] [root] Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/isaac-sim/GRUtopia/grutopia/core/task/task.py", line 77, in get_observations
    obs[robot_name] = robot.get_obs()
  File "/isaac-sim/GRUtopia/grutopia_extension/robots/humanoid.py", line 251, in get_obs
    obs[c_obs_name] = controller_obs.get_obs()
  File "/isaac-sim/GRUtopia/grutopia_extension/controllers/web_chat_box_controller.py", line 44, in get_obs
    chat_control = get_chat_control()
  File "/isaac-sim/GRUtopia/grutopia/core/datahub/web_ui_api.py", line 125, in get_chat_control
    ok, json_data = AsyncRequest.get(uuid, GetChatControlUrl)
  File "/isaac-sim/GRUtopia/grutopia/core/util/async_req.py", line 69, in get
    res = cls.res_info[key].json()
  File "/isaac-sim/.venv/lib/python3.10/site-packages/httpx/_models.py", line 756, in json
    return jsonlib.loads(self.text, **kwargs)
  File "/isaac-sim/kit/python/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/isaac-sim/kit/python/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/isaac-sim/kit/python/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

HanqingWangAI, 2024-07-22:
Thank you for your interest in our project and for the valuable feedback! It appears there is an issue with the unexpected format of the LLM's output. We will investigate this issue promptly and update you in this post once it is resolved.

uzuku, 2024-07-22:
@qZhang88 It seems the exception is caused by a bad response from WebUI API, which is expected to be a valid JSON. Could you run the following command when the exception is raised and provide the output here?

```
curl 127.0.0.1:9000/api/grutopia/getChatList
```

qZhang88, 2024-07-22:
> curl 127.0.0.1:9000/api/grutopia/getChatList

no response

```
(.venv) root@gpu-server:/isaac-sim# curl 127.0.0.1:9000/api/grutopia/getChatList
(.venv) root@gpu-server:/isaac-sim#
```

qZhang88, 2024-07-22:
a first seems to a proxy problem

```
(.venv) root@gpu-server:/isaac-sim# curl -v 127.0.0.1:9000/api/grutopia/getChatList
* Uses proxy env variable http_proxy == 'http://10.1.0.176:7890'
*   Trying 10.1.0.176:7890...
* Connected to (nil) (10.1.0.176) port 7890 (#0)
> GET http://127.0.0.1:9000/api/grutopia/getChatList HTTP/1.1
> Host: 127.0.0.1:9000
> User-Agent: curl/7.81.0
> Accept: */*
> Proxy-Connection: Keep-Alive
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 502 Bad Gateway
< Connection: keep-alive
< Keep-Alive: timeout=4
< Proxy-Connection: keep-alive
< Content-Length: 0
< 
* Connection #0 to host (nil) left intact
```

but second try after rm proxy, gives
```
(.venv) root@gpu-server:/isaac-sim# curl -v 127.0.0.1:9000/api/grutopia/getChatList
*   Trying 127.0.0.1:9000...
* connect to 127.0.0.1 port 9000 failed: Connection refused
* Failed to connect to 127.0.0.1 port 9000 after 0 ms: Connection refused
* Closing connection 0
curl: (7) Failed to connect to 127.0.0.1 port 9000 after 0 ms: Connection refused
```

APX103, 2024-07-23:
Hi. Please try this code after entering the container to get more error info:

``` sh
python ./webui/main.py
```

I think it is caused by the **backend of webui** not starting

qZhang88, 2024-07-23:
The error still exist.

One thing you should know, IP address is set to `0.0.0.0` so I can visit the from remote. And actually I could visit the webui from remote browser.

[image]

BTW, `python ./GRUtopia/demo/h1_locomotion.py  # start simulation` seems to work fine, while `python ./GRUtopia/demo/h1_house.py` failed

uzuku, 2024-07-24:
@qZhang88 I'll explain the cases briefly.

> a first seems to a proxy problem

The `JSONDecodeError` exception is caused by the empty response returned by the proxy.
If a valid json is returned, or the connection is refused, then the exception won't be raised. Currently the exception only occurs when HTTP connection can be established but a non-json response is returned (proxy server, in your case), which is a bug, and will be fixed soon.

> BTW, `python ./GRUtopia/demo/h1_locomotion.py` seems to work fine, while `python ./GRUtopia/demo/h1_house.py` failed

This works as expected, as h1_house demo declares `web_chat` controller in its setting (see `demo/configs/h1_house.yaml`) while h1_locomotion does not, and the HTTP request is made by the `web_chat` controller. You can remove the `web_chat` controller in h1_house.yaml and there will be no more HTTP stuff.

> One thing you should know, IP address is set to `0.0.0.0` so I can visit the from remote. And actually I could visit the webui from remote browser.

This works as expected too, a bit weird  though. The WebRTC UI is provided by Isaac Sim itself, and listen to 0.0.0.0 by default. A more featured WebUI is provided in GRUtopia to communicate with NPC, which have the WebRTC UI as a part of it, and it depends the address specified by the `WEBUI_HOST` to show the WebRTC properly. This is not so intuitive though, and we may polish it in the future.

APX103, 2024-07-26:
Hello, I closed this issue. We can reopen it if necessary.

closed 2024-07-26 by APX103

1018weijia, 2024-08-29:
I also encountered this error while running the h1_npc.py. How can I solve it?