InternRobotics/InternUtopia #4: Running lag

fighting-j, 2024-07-23:
Hello, when I run python ./GRUtopia/demo/h1_house.py according to the instructions, the environment loads very slowly, the following warning appears：
2024-07-23 07:50:21 [15,953ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-07-23 07:50:21 [15,953ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-07-23 07:50:21 [15,954ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-07-23 07:50:21 [15,954ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-07-23 07:50:21 [15,954ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2024-07-23 07:50:21 [15,954ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-07-23 07:50:22 [16,540ms] [Error] [rtx.neuraylib.plugin] Cannot find `Documents/Kit/shared` directory.
2024-07-23 07:50:26 [20,614ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-07-23 07:50:26 [20,615ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
...
2024-07-23 07:59:29 [283,180ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-07-23 07:59:29 [283,180ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-07-23 07:59:29 [283,180ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-07-23 07:59:29 [283,180ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-07-23 07:59:29 [283,180ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.

And the robot is very slow to operate after entering the environment.But my graphics card usage is not full.
[image]
I want to know if there is any way to make the environment smoother.

HanqingWangAI, 2024-07-23:
Thank you for your interest in our project. Could you let us know the FPS you get when running the demo? You can enable the FPS display by clicking the eye icon in the GUI.
[image]

fighting-j, 2024-07-23:
[image]
This is the FPS I was running at.

HanqingWangAI, 2024-07-23:
Hi @fighting-j, I ran this demo on an RTX 4090 device, achieving about 17 FPS. It appears there may be another Python process running on your GPU, which could be affecting performance. Please check for any such processes.
[image]
Additionally, you can increase the FPS by adding `rendering_interval: 1` under this line:
https://github.com/OpenRobotLab/GRUtopia/blob/1078eb49b02f90260af1fe28518799f6f1bac1d5/demo/configs/h1_house.yaml#L3
[image]

fighting-j, 2024-07-24:
I added rendering_interval: 1 as you suggested, and made sure there were no other programs running, and the FPS improved a bit, but it's still not at your level.
[image]
[image]
Here are some warnings that appear when I run it. Will it have any impact?
[image]
[image]

HanqingWangAI, 2024-07-24:
Hi @fighting-j, can you also check the memory usage and the CPU utils? I paste the process state here FYI.
[image]
One more thing you can do to improve FPS is to disable the cameras on the H1.
[image]
[image]

fighting-j, 2024-07-24:
[image]
[image]
The above is the running situation after I removed the camera, and there is still a big gap between your results and mine.

HanqingWangAI, 2024-07-24:
@fighting-j, it seems that a Java process is heavily utilizing your CPU, resulting in significantly lower CPU usage (300~ compared to 1000~) for the demo process. Can you terminate the java program and run the demo again?

fighting-j, 2024-07-24:
Indeed, I found that there was a java process occupying the CPU. After I ended it, the frame rate is now normal. Thank you for your help.

HanqingWangAI, 2024-07-24:
Thank you for your feedback. This issue will be closed as it is resolved.

closed 2024-07-24 by HanqingWangAI