InternRobotics/InternUtopia #5: Omniverse Streaming Client

qZhang88, 2024-07-23:
Is it possible to use Omniverse Streaming Client to connect sim in docker.

[image]

APX103, 2024-07-24:
try this script `/isaac-sim/isaac-sim.headless.native.sh` with container's `--network host`

APX103, 2024-07-24:
try this script `./isaac-sim.headless.native.sh --allow-root` at `/isaac-sim` in container. with container flag `--network host`.

qZhang88, 2024-07-24:
> try this script `./isaac-sim.headless.native.sh --allow-root` at `/isaac-sim` in container. with container flag `--network host`.

directly run script is ok, but I want to connect to those demos.

APX103, 2024-07-24:
Well. This feature is not supported yet. It well be supported soon.
I pushed a new demo you can try: [Here is the commit info](https://github.com/OpenRobotLab/GRUtopia/commit/39c043b606f5ed3cf1a82bffa7f17d07eeb08c35)

try run this in container:

``` sh
python ./GRUtopia/demo/h1_locomotion_native.py
```

APX103, 2024-07-26:
Hello, I closed this issue. We can reopen it if necessary.

closed 2024-07-26 by APX103