InternRobotics/InternUtopia #6: What is the reason for displaying network errors in the webui interface?

1018weijia, 2024-07-24:
The interface I encountered is as follows. Where did the problem occur?

[image]

HanqingWangAI, 2024-07-25:
Hi @1018weijia, thank you for your interest in our project! Could @APX103 please check this issue?

APX103, 2024-07-25:
This issue is most likely caused by the browser not being able to connect to the backend program.
There might be 2 reasons:
1. `webui_start.sh` start the backend of webui failed.
2. The backend program port is unreachable

for debugging，try `python ./webui/main.py` when entering the container for more details~

ykx3, 2024-07-29:
hello, I also encountered this problem and the log:
[image] 
how can I change the port?

APX103, 2024-07-29:
> hello, I also encountered this problem and the log: ![截图 2024-07-29 15-53-36](https://private-user-images.githubusercontent.com/154034514/352962903-fabc8a43-50d9-4cd8-a1cc-b8ab7d8b0f07.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyNDk1MTUsIm5iZiI6MTcyMjI0OTIxNSwicGF0aCI6Ii8xNTQwMzQ1MTQvMzUyOTYyOTAzLWZhYmM4YTQzLTUwZDktNGNkOC1hMWNjLWI4YWI3ZDhiMGYwNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyOVQxMDMzMzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMmUyMmJjY2VhZWQyMTcxMTUxMjQ3MzEwYWRhNzk4MjIyYjNkYmQ3Mzk2Nzc1MTk3MmExNDVjZmRhZGNlNWFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7uUtiGcCHsDCSugInPg7-cD11zgaaX0RkGV8EV1WX1s) how can I change the port?

Hi @ykx3 , Sorry there is no easy way to change the port (coming soon...), you can try this:

1. Add port param. Replace this line with `nohup python ./webui/main.py --port={your_port} &> webui.log &`
https://github.com/OpenRobotLab/GRUtopia/blob/6e20ccdff67faa4a485c93528b925829477a6ffb/webui_start.sh#L15
2. Change nginx config. Replace this line with `proxy_pass         http://127.0.0.1:{your_port}/api/;`
https://github.com/OpenRobotLab/GRUtopia/blob/6e20ccdff67faa4a485c93528b925829477a6ffb/webui/fe/nginx/default#L12

Hope this helps~

ykx3, 2024-07-30:
https://github.com/OpenRobotLab/GRUtopia/issues/6#issuecomment-2255590906 Thank you for your help and I have solved the problem by killing the process using this port

closed 2024-08-11 by 1018weijia