InternRobotics/InternUtopia #9: NPC in local?

mialana, 2024-07-26:
您好。I am curious if there are plans to eventually make NPC interaction available in the local GRUtopia environment? Thanks!

Boyu-Mi, 2024-07-26:
Thanks for your interest. The code of LLM inference is at https://github.com/OpenRobotLab/GRUtopia/blob/6e20ccdff67faa4a485c93528b925829477a6ffb/grutopia/npc/llm_caller.py#L82. You can modify the `api_base_url`if you are using an local llm server or replace it with inference code of your local LLM.

mialana, 2024-07-26:
Ok, thank you. And if I want to host the WebUI without using Docker, is there a quick way to do this?

uzuku, 2024-07-26:
> And if I want to host the WebUI without using Docker, is there a quick way to do this?

@mialana Currently there's no quick way to realize that, since nginx is required to serve the WebUI frontend, and we believe that setting it up with docker is a simpler way for most of the researchers. You can refer to [Dockerfile](https://github.com/OpenRobotLab/GRUtopia/blob/main/Dockerfile) for the the settings if a local nginx is more suitable for you.

By the way, the WebUI process is able to communicate to locally installed Isaac Sim, but some modification is necessary to make WebRTC work properly:

https://github.com/OpenRobotLab/GRUtopia/blob/6e20ccdff67faa4a485c93528b925829477a6ffb/Dockerfile#L16-L18

You can do it against the locally installed Isaac Sim, and then start WebUI process only in the docker container. They should be able to work with each other.

closed 2024-08-07 by Boyu-Mi