InternRobotics/InternUtopia #13: Unable to find target for this triple (no targets are registered) Fatal Python error: Aborted

1018weijia, 2024-08-13:
[image]
After running h1_locomotion. py, the isaac sim window appears and crashes after seeing half of the robots. The output record is as shown above. Could you tell me what is the reason for this?

uzuku, 2024-08-21:
It seems like the Isaac Sim installation is broken on the machine. Can you provide your Isaac Sim version and platform setting (OS, version, hardware spec, etc) ? 

And can you try the Isaac Sim standalone example to see if it can be launched successfully? Please provide the output if any error occurs.

```python
$ cd PATH/TO/ISAAC_SIM/ROOT
$ ./python.sh standalone_examples/api/omni.isaac.franka/follow_target_with_ik.py
```

1018weijia, 2024-08-23:
> It seems like the Isaac Sim installation is broken on the machine. Can you provide your Isaac Sim version and platform setting (OS, version, hardware spec, etc) ?
>
> And can you try the Isaac Sim [...]

I deployed on a Linux system using 2 * RTX 4070, with a minimum driver adaptation of 535. I noticed that the recommended driver for this project is 525, and I suspect it may be due to a driver mismatch. I tried your suggestion and found that it still couldn't run, most likely due to a driver issue, which is very frustrating.

1018weijia, 2024-08-23:
> I deployed on a Linux system using 2 * RTX 4070, with a minimum driver adaptation of 535. I noticed that the recommended driver for this project is 525, and I suspect it may be due to a driver [...]

Just to add, I am using Ubuntu version 22.04