InternRobotics/InternUtopia #14: May I ask if you can provide a Windows version of the project installation? Thank you.

1018weijia, 2024-08-16:


uzuku, 2024-08-21:
Technically if you have a ready Isaac Sim installation on Windows and a corresponding python environment, you can install the project by just

```shell
cd PATH/TO/ISAAC_SIM/ROOT
git clone git@github.com:OpenRobotLab/GRUtopia.git
cd GRUtopia
pip install -e .
```

For the python environment setup on Windows you can refer to https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html

1018weijia, 2024-08-23:
> Technically if you have a ready Isaac Sim installation on Windows and a corresponding python environment, you can install the project by just
>
> ```shell
> cd PATH/TO/ISAAC_SIM/ROOT
> git clone [...]

Thank you for your answer. I have successfully deployed on Windows and run three of the demos. Only the last NPC demo has not been run yet. Additionally, the sh files used in the project also need to be converted to bat files to run on Windows

closed 2024-08-23 by 1018weijia

1018weijia, 2024-08-24:
The replication of Windows only requires converting the sh file into a bat file for execution. The other steps are the same as local Linux installation. The bat file can be found on the website:
https://www.notion.so/Windows-support-for-GRUtopia-ca3a286b7ab44c57b8538582ad6267a3?pvs=4