InternRobotics/InternUtopia #15: 是否支持替换成自己的场景？

Yu-xm, 2024-08-26:
是否支持替换成自己的场景？

HanqingWangAI, 2024-08-27:
Hi @Yu-xm , you can replace this line https://github.com/OpenRobotLab/GRUtopia/blob/6e20ccdff67faa4a485c93528b925829477a6ffb/demo/configs/h1_house.yaml#L8 with the path to your scene assets. Please note that the scene must be bound with colliders to allow the robots to navigate within it.

RongXingDaMoWang, 2024-08-30:
替换自己的场景后，自己场景的注释怎么添加

HanqingWangAI, 2024-08-30:
@RongXingDaMoWang , 没有理解你的意思，你的注释是指什么？提供更多的信息有助于我们理解你的需求。

RongXingDaMoWang, 2024-09-01:
@HanqingWangAI ，您好，我的意思是，github上的场景中各个物体区域都是有注释信息的，要是我提供新的场景，新的场景中的各个物体的注释信息怎么添加？

HanqingWangAI, 2024-09-02:
@RongXingDaMoWang 目前我们提供的场景是按照scope->category->instance(->part)的层级进行组织的，物体级别的物体划分已经在结构中体现了。场景的语言描述等标注将会在下一次release中发布，这些自然语言的标注信息在外部的json文件里，不是嵌入在usd文件中的。对于新的场景可以使用[USD Composer](https://docs.omniverse.nvidia.com/composer/latest/index.html)或者OpenUSD的API根据你的标注信息来调整你的场景结构。