InternRobotics/InternUtopia #16: PhysX Shape failed to be created on a prim:

RongXingDaMoWang, 2024-08-27:
您好，感谢你们的分享，我在在运行城市和房屋的demo的时候，提示如下问题，希望能够得到解决，感谢
[image]

[image]

报错内容：
2024-08-27 08:03:55  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE /buildAgent/work/eb2f45c4acc808a0/physx/source/physx/src/NpFactory.cpp, LINE 863

2024-08-27 08:03:55  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE /buildAgent/work/eb2f45c4acc808a0/physx/source/physx/src/NpFactory.cpp, LINE 863

RongXingDaMoWang, 2024-08-27:
补充：
2024-08-27 08:08:50  [Error] [omni.physx.plugin] Unable to create convex mesh for:/World/env_0/scene/Xform/Building/home/Meshes/Animation/electriccooker/SM_01_ZFE3YERVAIJH2PTUKA888888/Group_140/Component_4

2024-08-27 08:06:47  [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: file:/home/amax/.local/share/ov/pkg/isaac-sim-2023.1.1/GRUtopia/assets/scenes/demo_city/half/Materials/AM234_020_Railing.mdl?watch=00007fe6cd975280(94,5): C152 'VRayMtl' : cannot convert argument 'texmap_bump' from 'struct ::base::texture_return' to 'float3'

HanqingWangAI, 2024-08-30:
Hi @RongXingDaMoWang, thank you for your feedback.  Can you paste the complete traceback of Errors so we can look into this issue?

percypeng5221, 2024-08-31:
I think I'm experiencing the same error when excuting h1_house.py? @HanqingWangAI The complete error is here: 

`Starting kit application with the following args:  ['/home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/exts', '--ext-folder', '/home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20240831_123808.log
2024-08-31 16:38:08 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.027s] [ext: omni.kit.async_engine-0.0.0] startup
[0.181s] [ext: omni.activity.core-1.0.1] startup
[0.183s] [ext: omni.assets.plugins-0.0.0] startup
[0.183s] [ext: omni.stats-0.0.0] startup
[0.184s] [ext: omni.client-1.0.1] startup
[0.188s] [ext: omni.activity.profiler-1.0.2] startup
[0.189s] [ext: omni.gpu_foundation-0.0.0] startup
[0.194s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.194s] [ext: carb.windowing.plugins-1.0.0] startup
[0.201s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.28.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090 Laptop.. | Yes: 0 |     | 16376   MB | 10de      | 0          |
|     |                                  |        |     |            | 2757      | b7ca93e3.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.8.0-40-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900HX | Cores: 24 | Logical: 48
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31895 | Free Memory: 25864
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[1.000s] [ext: omni.kit.pipapi-0.0.0] startup
[1.001s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.001s] [ext: omni.pip.compute-1.2.0] startup
[1.001s] [ext: omni.pip.cloud-1.0.1] startup
[1.002s] [ext: omni.isaac.core_archive-2.2.1] startup
[1.002s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[1.023s] [ext: omni.kit.telemetry-0.5.0] startup
[2024-08-31 12:38:09,292][INFO] /home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/exts/omni.kit.telemetry/omni/kit/telemetry/impl/sentry_extension.py[line:78] -: sentry is disabled for external build
[2024-08-31 12:38:09,293][INFO] /home/percy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/exts/omni.kit.telemetry/omni/kit/telemetry/impl/sentry_extension.py[line:169] -: sentry is disabled for external build
[1.036s] [ext: omni.kit.loop-isaac-1.1.0] startup
[1.037s] [ext: omni.kit.test-0.0.0] startup
[1.038s] [ext: omni.isaac.ml_archive-1.1.3] startup
[1.038s] [ext: omni.mtlx-0.1.0] startup
[1.039s] [ext: omni.usd.config-1.0.3] startup
[1.041s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.041s] [ext: omni.usd.libs-1.0.0] startup
[1.102s] [ext: omni.appwindow-1.1.5] startup
[1.105s] [ext: omni.kit.renderer.core-0.0.0] startup
[1.172s] [ext: omni.kit.renderer.capture-0.0.0] startup
[1.174s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[1.230s] [ext: carb.audio-0.1.0] startup
[1.232s] [ext: omni.ui-2.18.5] startup
[1.239s] [ext: omni.uiaudio-1.0.0] startup
[1.239s] [ext: omni.kit.mainwindow-1.0.1] startup
[1.240s] [ext: omni.kit.uiapp-0.0.0] startup
[1.240s] [ext: omni.usd.schema.physics-0.0.0] startup
[1.307s] [ext: omni.usd.schema.audio-0.0.0] startup
[1.310s] [ext: omni.usd.schema.forcefield-0.0.0] startup
[1.313s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[1.317s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[1.317s] [ext: omni.usd.schema.semantics-0.0.0] startup
[1.322s] [ext: omni.usd.schema.physx-0.0.0] startup
[1.336s] [ext: omni.usd.schema.anim-0.0.0] startup
[1.347s] [ext: omni.anim.navigation.schema-105.2.0] startup
[1.349s] [ext: omni.anim.graph.schema-105.2.2] startup
[... 425 lines cut ...]
[... cut ...]
2024-08-31 16:38:38 [29,968ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-08-31 16:38:38 [29,968ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
2024-08-31 16:38:38 [29,968ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL.
100
200`

RookieXwc, 2024-09-13:
@HanqingWangAI ，Hello, thank you for your great work. I have encountered the same problem as @percypeng5221 and @RongXingDaMoWang . May I ask what is going on?
[image]

RookieXwc, 2024-09-13:
https://forums.developer.nvidia.com/t/physx-error-supplied-pxgeometry-is-not-valid/260265
In this similar question, it is mentioned that this error typically occurs when there is a problem with the geometry of one or more of the collision shapes in your robot model. May I ask if the collision occurred in the code or if there is a problem with my running environment? @HanqingWangAI

HanqingWangAI, 2024-09-14:
Hi @RookieXwc @percypeng5221 @RongXingDaMoWang, we basically located the issue. It is caused by the in-compatible way of generating approximated collider for rigid bodies. We are still solving the errors. A temporal workaround is to disable all rigidbody in the scene and adopt `meshSimplification` for generating collider. You can achieve this by applying `bind_static` to the root prim.

```python

from pxr import Usd, UsdPhysics, UsdGeom

def set_collider_(mesh, approx=UsdPhysics.Tokens.convexDecomposition, init_state=True):
    """set collider for a mesh
    Args:
    - mesh: Usd.Prim -> Mesh
    - approx: Convex approximation method, e.g. ConvenDecomposition, ConvexHull
    - init_state: init state of collider(whether activate)
    """
    if mesh.GetTypeName() == "Mesh" and UsdPhysics.CollisionAPI.CanApply(mesh):
            collider = UsdPhysics.CollisionAPI.Apply(mesh)
            meshcollider = UsdPhysics.MeshCollisionAPI.Apply(mesh)            
            meshcollider.GetApproximationAttr().Set(approx)            
            collider.GetCollisionEnabledAttr().Set(init_state)

def set_collider(item, approx=UsdPhysics.Tokens.convexDecomposition, init_state=True):
    """set collider for all meshes of an Usd.Prim
    Args:
    - item: Usd.Prim, e.g. Xform
    """
    if len(item.GetChildren()) == 0:
        set_collider_(item, approx, init_state=init_state)
    for i in item.GetChildren():
        set_collider(i, approx, init_state=init_state)

def remove_collider_(mesh):
    if mesh.GetTypeName() == "Mesh":
        mesh.RemoveAPI(UsdPhysics.CollisionAPI)
        mesh.RemoveAPI(UsdPhysics.MeshCollisionAPI)

def remove_collider(item):
    if len(item.GetChildren()) == 0:
        remove_collider_(item)
    for i in item.GetChildren():
        remove_collider(i)

def remove_rigid_(prim):
    prim.RemoveAPI(UsdPhysics.RigidBodyAPI)
    if prim.IsA(UsdPhysics.PrismaticJoint) or prim.IsA(UsdPhysics.RevoluteJoint):
        attr = prim.GetAttribute("physics:jointEnabled")
        attr.Set(False)
    
def remove_rigid(item):
    remove_rigid_(item)
    for i in item.GetChildren():
        remove_rigid(i)
        
def bind_static(prim):
    remove_collider(prim)
    remove_rigid(prim)
    set_collider(prim, "meshSimplification")
```