InternRobotics/InternUtopia #19: question when trying tutorials

ykx3, 2024-09-02:
i follow the tutorial in [this](https://grutopia.github.io/tutorials/how-to-use-robot.html#usage), but i got a warning and it didn't work in my chorme. Could you give me some help?
```log
[48.224s] Simulation App Startup Complete
[48.252s] [ext: omni.kit.renderer.cuda_interop-1.0.1] startup
[48.263s] [ext: omni.kit.livestream.core-2.3.0] startup
[48.266s] [ext: omni.services.facilities.base-1.0.4] startup
[48.269s] [ext: omni.services.core-1.6.4] startup
[48.467s] [ext: omni.services.transport.server.base-1.1.1] startup
[48.470s] [ext: omni.kit.streamsdk.plugins-2.5.2] startup
[48.470s] [ext: omni.services.transport.server.http-1.3.0] startup
[2024-09-02 11:17:33,048][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:76] -: Looking for Python nodes to register in omni.services.transport.server.http-1.3.0
[2024-09-02 11:17:33,049][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:101] -:  -> Registered nodes from module omni.services.transport.server.http at /isaac-sim/extscache/omni.services.transport.server.http-1.3.0
[2024-09-02 11:17:33,049][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/registration.py[line:30] -: Registering nodes in /isaac-sim/extscache/omni.services.transport.server.http-1.3.0 imported as omni.services.transport.server.http
[2024-09-02 11:17:33,049][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:66] -: Registering Python Node Types from omni.services.transport.server.http at /isaac-sim/extscache/omni.services.transport.server.http-1.3.0 in omni.services.transport.server.http
[2024-09-02 11:17:33,049][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:67] -: ========================================================================================================================
[2024-09-02 11:17:33,050][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/extension_management.py[line:138] -: No dependency on omni.graph, therefore no nodes to register in omni.services.transport.server.http
[2024-09-02 11:17:33,050][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:72] -: ...None found, no registration to do
[2024-09-02 11:17:33,050][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:111] -: ...Skipping: No OmniGraph presence in the module omni.services.transport.server.http - No nodes in this module, do not remember it
[2024-09-02 11:17:33,050][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:120] -: OGN register omni.services.transport.server.http-1.3.0 took 1449085.000000
[48.511s] [ext: omni.kit.livestream.webrtc-2.3.2] startup

Failed to open [/var/run/utmp]
Active user not found. Using default user [kiosk]Streaming server started.
[2024-09-02 11:17:33,344][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:76] -: Looking for Python nodes to register in omni.kit.livestream.webrtc-2.3.2
[2024-09-02 11:17:33,345][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:101] -:  -> Registered nodes from module omni.kit.livestream.webrtc at /isaac-sim/extscache/omni.kit.livestream.webrtc-2.3.2+105.1.lx64.r.cp310
[2024-09-02 11:17:33,345][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/registration.py[line:30] -: Registering nodes in /isaac-sim/extscache/omni.kit.livestream.webrtc-2.3.2+105.1.lx64.r.cp310 imported as omni.kit.livestream.webrtc
[2024-09-02 11:17:33,346][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:66] -: Registering Python Node Types from omni.kit.livestream.webrtc at /isaac-sim/extscache/omni.kit.livestream.webrtc-2.3.2+105.1.lx64.r.cp310 in omni.kit.livestream.webrtc
[2024-09-02 11:17:33,346][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:67] -: ========================================================================================================================
[2024-09-02 11:17:33,349][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/extension_management.py[line:138] -: No dependency on omni.graph, therefore no nodes to register in omni.kit.livestream.webrtc
[2024-09-02 11:17:33,349][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:72] -: ...None found, no registration to do
[2024-09-02 11:17:33,349][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:111] -: ...Skipping: No OmniGraph presence in the module omni.kit.livestream.webrtc - No nodes in this module, do not remember it
[2024-09-02 11:17:33,350][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:120] -: OGN register omni.kit.livestream.webrtc-2.3.2 took 4897760.000000
[48.811s] [ext: omni.services.streaming.manager-0.3.10] startup
[2024-09-02 11:17:33,365][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:76] -: Looking for Python nodes to register in omni.services.streaming.manager-0.3.10
[2024-09-02 11:17:33,366][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:101] -:  -> Registered nodes from module omni.services.streaming.manager at /isaac-sim/extscache/omni.services.streaming.manager-0.3.10
[2024-09-02 11:17:33,366][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/registration.py[line:30] -: Registering nodes in /isaac-sim/extscache/omni.services.streaming.manager-0.3.10 imported as omni.services.streaming.manager
[2024-09-02 11:17:33,366][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:66] -: Registering Python Node Types from omni.services.streaming.manager at /isaac-sim/extscache/omni.services.streaming.manager-0.3.10 in omni.services.streaming.manager
[2024-09-02 11:17:33,366][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:67] -: ========================================================================================================================
[2024-09-02 11:17:33,367][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/extension_management.py[line:138] -: No dependency on omni.graph, therefore no nodes to register in omni.services.streaming.manager
[2024-09-02 11:17:33,367][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:72] -: ...None found, no registration to do
[2024-09-02 11:17:33,367][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:111] -: ...Skipping: No OmniGraph presence in the module omni.services.streaming.manager - No nodes in this module, do not remember it
[2024-09-02 11:17:33,367][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:120] -: OGN register omni.services.streaming.manager-0.3.10 took 1949666.000000
[48.829s] [ext: omni.services.streamclient.webrtc-1.3.8] startup
[2024-09-02 11:17:33,759][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:76] -: Looking for Python nodes to register in omni.services.streamclient.webrtc-1.3.8
[2024-09-02 11:17:33,760][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/extension.py[line:101] -:  -> Registered nodes from module omni.services.streamclient.webrtc at /isaac-sim/extscache/omni.services.streamclient.webrtc-1.3.8
[2024-09-02 11:17:33,760][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/registration.py[line:30] -: Registering nodes in /isaac-sim/extscache/omni.services.streamclient.webrtc-1.3.8 imported as omni.services.streamclient.webrtc
[2024-09-02 11:17:33,760][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:66] -: Registering Python Node Types from omni.services.streamclient.webrtc at /isaac-sim/extscache/omni.services.streamclient.webrtc-1.3.8 in omni.services.streamclient.webrtc
[2024-09-02 11:17:33,760][INFO] /isaac-sim/kit/exts/omni.graph/omni/graph/core/_impl/_registration/register_python_ogn.py[line:67] -: ========================================================================================================================
[... cut ...]
2024-09-02 11:23:26 [401,455ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-09-02 11:23:26 [401,455ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-09-02 11:23:26 [401,461ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-09-02 11:23:26 [401,639ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-09-02 11:23:26 [401,654ms] [Warning] [omni.core.ITypeFactory] Module /isaac-sim/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

HanqingWangAI, 2024-09-05:
@ykx3 , it appears that you can not access the omniverse nucleus server, which causes the `add_default_ground_plane` function to fail. We would consider deprecating this function in the next version. 

Is there any workaround available for this issue? @APX103

uzuku, 2024-09-05:
@ykx3 you can remove the `env` section from config file to disable the  behavior of `add_default_ground_plane `:

simulator:
  physics_dt: 1/240
  rendering_dt: 1/240

~~env:~~
  ~~bg_type: null~~

...

ykx3, 2024-09-05:
thank for your help

closed 2024-09-05 by ykx3