InternRobotics/InternUtopia #21: Open3D Dependency Issue in Docker Environment

ykx3, 2024-09-05:
I am encountering an issue when trying to use Open3D within the Docker environment. It appears that a library version is too low, causing problems with Open3D. I am unsure how to adjust the library versions within the Docker container.
```log
(.venv) root@admin:/isaac-sim# python 
Python 3.10.13 (main, Oct 18 2023, 20:21:26) [GCC 7.3.1 20180303 (Red Hat 7.3.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import open3d
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/isaac-sim/.venv/lib/python3.10/site-packages/open3d/__init__.py", line 112, in <module>
    import open3d.visualization
  File "/isaac-sim/.venv/lib/python3.10/site-packages/open3d/visualization/__init__.py", line 19, in <module>
    from .draw_plotly import draw_plotly
  File "/isaac-sim/.venv/lib/python3.10/site-packages/open3d/visualization/draw_plotly.py", line 12, in <module>
    from dash import html
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/__init__.py", line 5, in <module>
    from .dependencies import (  # noqa: F401,E402
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/dependencies.py", line 1, in <module>
    from dash.development.base_component import Component
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/development/__init__.py", line 1, in <module>
    from . import base_component  # noqa:F401
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/development/base_component.py", line 8, in <module>
    from .._utils import patch_collections_abc, stringify_id, OrderedSet
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/_utils.py", line 17, in <module>
    from dash.types import RendererHooks
  File "/isaac-sim/.venv/lib/python3.10/site-packages/dash/types.py", line 1, in <module>
    from typing_extensions import TypedDict, NotRequired
ImportError: cannot import name 'NotRequired' from 'typing_extensions' (/isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle/typing_extensions.py)
```
and when I trying to uninstall that lib, i got
```log
(.venv) root@admin:/isaac-sim# pip uninstall typing_extensions
Found existing installation: typing-extensions 3.10.0.2
Not uninstalling typing-extensions at /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle, outside environment /isaac-sim/.venv
Can't uninstall 'typing-extensions'. No files were found to uninstall.
```
could u give me some help?

uzuku, 2024-09-06:
@ykx3 The typing-extensions package is shipped with Isaac Sim pre-bundle and thus cannot be uninstalled within the virtual env. You can exit the virtual env and uninstall the package, then install open3d in the virtual env again.

```shell
(.venv) root@admin:/isaac-sim# deactivate
root@admin:/isaac-sim# ./python.sh -m pip uninstall typing-extensions
root@admin:/isaac-sim# source .venv/bin/activate
(.venv) root@admin:/isaac-sim# pip install open3d
```

However keep in mind that upgrading the version of package shipped with Isaac Sim may broke some Isaac Sim internal utility. You can try it out yourself and see if any required feature is broken in your case.

Currently we are migrating GRUtopia to newer version of Isaac Sim which is compatible with open3d, and we will release it when stable.

ykx3, 2024-09-06:
thank you, I've uninstalled it and it did work well.

closed 2024-09-06 by ykx3