InternRobotics/InternUtopia #23: How do I install Issacsim 2023.1.1

Decycle, 2024-09-27:
In the Nvidia Omniverse launcher I can no longer find the 2023.1.1 version of Issacsim, what should I do to install the correct version?

uzuku, 2024-10-10:
You may use the [container installation](https://grutopia.github.io/get_started/installation.html#install-with-docker-linux) of Isaac Sim 2023.1.1 if no workstation version is available in Omniverse Launcher. You can use streaming client with `native=True` to achieve a similar experience as workstation.

https://github.com/OpenRobotLab/GRUtopia/blob/aadfca80080f26e06360bb216d47f491553a271c/demo/h1_locomotion_native.py#L15

Currently we are working on migrating GRUtopia to Isaac Sim 4.x version, and will release it when stable.

uzuku, 2025-02-28:
Hi @Decycle Omniverse has provided the archive of old versions of Isaac Sim in their [doc](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/download.html#download-archive).

closed 2025-02-28 by uzuku