InternRobotics/InternUtopia #24: How to load this project with isaacsim's version being 4.2

RonaldZhou11, 2024-09-29:
Hi, I appreciate your work. My version of the isaacsim is 4.2. and I wonder how to load your project with this version(or 4.1). Could you give me some hint about where to modify?

uzuku, 2024-10-10:
Currently we are working on migrating GRUtopia to Isaac Sim 4.x version, majoring at fixing some stability issues.
We will release it once stable.

uzuku, 2025-02-24:
Hi @Little-Sayu The current main branch is compatible with Isaac Sim 4.x.

Jacking91, 2025-02-24:
> Hi [@Little-Sayu](https://github.com/Little-Sayu) The current main branch is compatible with Isaac Sim 4.x.

Hi, so it also supports Isaac Sim 4.5?

uzuku, 2025-02-28:
> Hi, so it also supports Isaac Sim 4.5?

Hi @Jacking91  We are working on it, and you can track it at https://github.com/OpenRobotLab/GRUtopia/issues/29 .

closed 2025-02-28 by uzuku