InternRobotics/InternUtopia #26: A control problem about the H1 robot's stepping

1018weijia, 2024-12-31:
I found that the H1 robot keeps stepping in your scene, which causes its original position to constantly shift? May I ask how to solve this?

HanqingWangAI, 2025-01-16:
Hi @1018weijia. The shift is unavoidable with the current locomotion policy. We plan to update the policy to incorporate a stable standing behavior when no command is provided. As a temporary solution, you can customize the controller to mitigate the shift error by issuing a reverse velocity command whenever the error exceeds the defined threshold.

uzuku, 2025-03-18:
Hi @1018weijia You can use [G1](https://github.com/OpenRobotLab/GRUtopia/blob/main/grutopia/demo/g1_locomotion.py) instead, which can stand stable when no command is provided.

closed 2025-03-18 by uzuku