InternRobotics/InternUtopia #28: custom differential drive wheel-robot

superjuices, 2025-02-24:
Hi, great work! How do I add a custom differential drive car，I have the urdf and usd of this wheel-robot.

uzuku, 2025-02-24:
Hi @superjuices You can refer to [how-to-add-robot](https://grutopia.github.io/advanced_tutorials/how-to-add-robot.html) to create your custom robot, and [how-to-add-controller](https://grutopia.github.io/advanced_tutorials/how-to-add-controller.html) to create a differential controller for it to move around.

We'll add a wheeled robot (jetbot) example later in this week, and you can take it as a reference.

superjuices, 2025-02-24:
> Hi [@superjuices](https://github.com/superjuices) You can refer to [how-to-add-robot](https://grutopia.github.io/advanced_tutorials/how-to-add-robot.html) to create your custom robot, and [...]
Thank you very much! I will look at the document and try to learn. I am looking forward to your example. It will be better for me to get started with this example.

superjuices, 2025-03-03:
Think you！I will give it a try

closed 2025-03-18 by uzuku