InternRobotics/InternUtopia #29: Can this project run on isaac-sim4.5.0?

MrCapricornLiu, 2025-02-25:


uzuku, 2025-02-25:
Hi @MrCapricornLiu . Currently we are working on examining all our features on Isaac Sim 4.5.0.
Until now we can confirm that the main features are working well, and once all the examinations are completed we'll update our doc to officially support 4.5.0. It won't take long.

uzuku, 2025-03-18:
Hi @MrCapricornLiu We've completed all the checks and unfortunately we found that the GRScenes-100 cannot be loaded properly in Isaac Sim 4.5. It may take a considerable efforts to resolve it, and we are working on it.

chlxydl, 2025-05-26:
Hi, I would like to ask weather there is any update for supporting to Isaacsim 4.5.0?

ruziniuuuuu, 2025-06-01:
Could you guys support isaacsim 4.5 in the version 2.1.0 now?

luorui93, 2025-07-08:
> Hi [@MrCapricornLiu](https://github.com/MrCapricornLiu) We've completed all the checks and unfortunately we found that the GRScenes-100 cannot be loaded properly in Isaac Sim 4.5. It may take a considerable efforts to resolve it, and we are working on it.

May I know what's the problem you encountered when loading GRScenes-100 in IsaacSim 4.5? I tried it myself and found that it took extremely long to load a scene (>5mins). Was this the same problem you had before?