InternRobotics/InternUtopia #32: Asserts Download Error.

BJHYZJ, 2025-02-28:
(.venv) root@yanzj-System-Product-Name:/isaac-sim# python -m grutopia.download_assets
Welcome to the Dataset Downloader!
Please choose the dataset source for download (huggingface/modelscope/openxlab): openxlab

Options:
    - min (~500MiB): only a minimum set of assets will be downloaded, without the GRScenes-100 and assets required by benchmark baselines.
    - full: (~80GiB to download, ~230GiB after unzipped): full assets will be downloaded.
Please choose (min/full): full
Asset (full) will be installed under this location: /isaac-sim/GRUtopia/grutopia/assets
If you want to use a different one, please type in (must be absolute path): 
Do you want to continue? (Y/n): Y
Creating Conda environment 'grutopia_asset__5HHqWvbF' for downloading dataset...
Traceback (most recent call last):
  File "/isaac-sim/kit/python/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/isaac-sim/kit/python/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/isaac-sim/GRUtopia/grutopia/download_assets.py", line 304, in <module>
    main()
  File "/isaac-sim/GRUtopia/grutopia/download_assets.py", line 278, in main
    create_conda_env()
  File "/isaac-sim/GRUtopia/grutopia/download_assets.py", line 37, in create_conda_env
    subprocess.check_call(['conda', 'create', '--name', DEFAULT_ENV_NAME, '--yes', 'python=3.10'])
  File "/isaac-sim/kit/python/lib/python3.10/subprocess.py", line 364, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/isaac-sim/kit/python/lib/python3.10/subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/isaac-sim/kit/python/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/isaac-sim/kit/python/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'conda'

uzuku, 2025-02-28:
Hi @BJHYZJ  We're fixing this and will release ASAP.

uzuku, 2025-02-28:
Hi @BJHYZJ We've fixed assets downloading in Docker. Please use the latest main branch to rebuild the image. Feel free to reopen this issue if you have further problems.

closed 2025-02-28 by uzuku

BJHYZJ, 2025-02-28:
Very efficient, thank you very much!