InternRobotics/InternUtopia #35: Running Benchmark Baseline Error.

BJHYZJ, 2025-03-01:
Thank you very much for such a good open source. I tried to follow the tutorial given by Run Benchmark Baseline and found some problems with running and conda environment configuration. I think there are some bugs in the documentation. Can you help test whether it can actually run and write the environment configuration clearly (it seems that there are multiple environments). In short, can you test the feasibility of the benchmark and improve the documentation so that it is easier to follow and reproduce better.

HanqingWangAI, 2025-03-03:
Hi @BJHYZJ , thank you for the feedback. We are looking into this issue and improving the documentation.

uzuku, 2025-03-10:
Hi @BJHYZJ We've solved some issues on running the benchmark baseline and polish the document about it. You can try it out with the latest main branch and doc: https://grutopia.github.io/get_started/run-benchmark-baseline.html.

If you have further problems, feel free to report it back.

closed 2025-03-18 by uzuku