InternRobotics/InternUtopia #36: Memory Consumption During play_scene.py Execution

hany01rye, 2025-03-03:
Hello! I’m facing an issue while running the Python script `warmup.py` & `play_scene.py` on my machine which has 32GB of RAM. The memory usage keeps increasing until my computer becomes unresponsive and freezes.

[image]
[image]
[image]

tianshihan818, 2025-03-04:
@hany01rye Hello! Thanks for your feedback. We tested on several scenes, here are indeed cases where the memory cost exceeds 32G. Actually, it normally cost up to 50G memory in average. 
We will working on the resource cost optimization.

hany01rye, 2025-03-04:
> [@hany01rye](https://github.com/hany01rye) Hello! Thanks for your feedback. We tested on several scenes, here are indeed cases where the memory cost exceeds 32G. Actually, it normally cost up to 50G memory in average. We will working on the resource cost optimization.

OKOK.
Thanks so much for helping to resolve this issue.

closed 2025-03-04 by hany01rye