InternRobotics/InternUtopia #38: MDL Material Loading Error in h1_social_navigation.py Execution

c99z, 2025-03-04:
Hello, I encountered an issue while running h1_social_navigation.py in my environment. The script fails to load MDL materials, causing rendering errors. Below is the error log I received:

```2025-03-04 06:00:19 [12,074,489ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (OmniUsdMdl): ::Z73file_3A::home::dell::GRUtopia::grutopia::assets::OpenRobotLab___GRScenes::scenes::Z60GRScenes_2D100::home_scenes::scenes::MV7J6NIKTKJZ2AABAAAAADY8_usd::models::object::articulated::electriccooker::Z99_3441837a19229a98d827c1df74b3f0207::Materials::MI_6577347511bd150001eb1e49
2025-03-04 06:00:19 [12,074,489ms] [Error] [omni.usd] USD_MDL: in LoadModule at line 247 of ../../source/plugins/usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for '/home/dell/GRUtopia/grutopia/assets/OpenRobotLab___GRScenes/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADY8_usd/models/object/articulated/electriccooker/441837a19229a98d827c1df74b3f0207/Materials/MI_6577347511bd150001eb1e49.mdl' is Invalid

2025-03-04 06:00:19 [12,074,490ms] [Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 178 of ../../source/plugins/usdMdl/moduleRegistry.cpp -- Module: '/home/dell/GRUtopia/grutopia/assets/OpenRobotLab___GRScenes/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADY8_usd/models/object/articulated/electriccooker/441837a19229a98d827c1df74b3f0207/Materials/MI_6577347511bd150001eb1e49.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.

2025-03-04 06:00:20 [12,074,569ms] [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (rtx_scope0) failed: ::Z73file_3A::home::dell::GRUtopia::grutopia::assets::OpenRobotLab___GRScenes::scenes::Z60GRScenes_2D100::home_scenes::scenes::MV7J6NIKTKJZ2AABAAAAADY8_usd::models::object::articulated::electriccooker::Z99_3441837a19229a98d827c1df74b3f0207::Materials::MI_6577347511bd150001eb1e49
2025-03-04 06:00:20 [12,074,569ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (rtx_scope0): ::Z73file_3A::home::dell::GRUtopia::grutopia::assets::OpenRobotLab___GRScenes::scenes::Z60GRScenes_2D100::home_scenes::scenes::MV7J6NIKTKJZ2AABAAAAADY8_usd::models::object::articulated::electriccooker::Z99_3441837a19229a98d827c1df74b3f0207::Materials::MI_6577347511bd150001eb1e49
2025-03-04 06:00:20 [12,074,569ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/World/env_0/scene/Meshes/Animation/electriccooker/model_441837a19229a98d827c1df74b3f0207_0/Looks/MI_6577347511bd150001eb1e49/Shader'. createMdlModule failed.

```
**The above error appears many times, and it seems that many MDL materials failed to load.**
- Operating System: (Ubuntu 22.04)
- Isaac sim Version: (4.2.0)
- NVIDIA GPU Model (Nvidia Quadro RTX 6000)
The program seems to run normally, but it takes a long time.
How can I solve this problem?

tianshihan818, 2025-03-04:
@c99z  Hi! Did you [configure the environment variable `MDL_SYSTEM_PATH`](https://github.com/OpenRobotLab/GRUtopia/blob/main/toolkits/grscenes_scripts/README.md#configure-mdl-material-search-path)? I think the value in your case is `/home/dell/GRUtopia/grutopia/assets/OpenRobotLab___GRScenes/scenes/GRScenes-100/home_scenes/Materials`.

c99z, 2025-03-04:
> [@c99z](https://github.com/c99z) Hi! Did you [configure the environment variable [...]

 Thank you for your reply, I've fixed this issue but another problem has arisen：
```2025-03-04 08:11:24 [637,207ms] [Error] [omni.usd] USD_MDL: in LoadModule at line 247 of ../../source/plugins/usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for '/home/dell/GRUtopia/grutopia/assets/OpenRobotLab___GRScenes/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADY8_usd/Materials/DayMaterial.mdl' is Invalid

2025-03-04 08:11:24 [637,207ms] [Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 178 of ../../source/plugins/usdMdl/moduleRegistry.cpp -- Module: '/home/dell/GRUtopia/grutopia/assets/OpenRobotLab___GRScenes/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADY8_usd/Materials/DayMaterial.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.
```
It's about ```DayMaterial.mdl```
I've noticed that a similar issue has been raised, but I'm not sure if the bug has been fixed and that the solution to that problem doesn't apply directly to my problem。How can I fix it? Thank you very much!

RenieWell, 2025-03-31:
Same problem here

m-muaz, 2025-04-02:
Got same issue.

QinFFF, 2025-06-24:
Got same issue.

[image]

And some materials are red

[image]

tianshihan818, 2025-07-10:
> Got same issue.
> 
> [image]
> 
> And some materials are red
> 
> [image]

@QinFFF Hello! Thanks for your feedbacks. It seems like the materials symlinks are broken. You could reference this [issue](https://github.com/OpenRobotLab/GRUtopia/issues/65) to check whether your assets directory is organized correctly.