InternRobotics/InternUtopia #41: Bug when rendering the home scenes

MiangChen, 2025-03-09:
Ubuntu 20.04
IsaacSim 4.2.0

The table_scene.usd can be rendered properly:
```
python toolkits/grscenes_scripts/play_scene.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd
```
[image]

However, when I try the scene in the commercial and home scenes, there is nothing:
```
 python toolkits/grscenes_scripts/play_scene.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADA8_usd/start_result_raw.usd
```

[image]

and the error occurs:
```
2025-03-09 14:33:20 [7,087ms] [Error] [omni.usd] USD_MDL: in LoadModule at line 247 of ../../source/plugins/usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for 'Materials/DayMaterial.mdl' is Invalid

2025-03-09 14:33:20 [7,087ms] [Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 178 of ../../source/plugins/usdMdl/moduleRegistry.cpp -- Module: 'Materials/DayMaterial.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.

2025-03-09 14:33:20 [7,087ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/Root/Looks/DayMaterial/DayMaterial'. Empty identifier: ''         and/or subIdentifier: ''
```

MiangChen, 2025-03-09:
I have try the method in https://github.com/OpenRobotLab/GRUtopia/issues/30#issuecomment-2690924097
```
from pxr import Usd

stage = Usd.Stage.Open("/home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADA8_usd/start_result_raw.usd")
hdr_prim = stage.GetPrimAtPath("/Root/__default_setting/HDR_Sphere")
hdr_prim.SetActive(False)
stage.GetRootLayer().Save()
```

[image]

the error disappers, but there is still nothing:

[image]

HanqingWangAI, 2025-03-09:
Hi @MiangChen, the black screen might be caused by the poor pose of the perspective camera. You may expand the stage panel on the right, select an object, and then press the F key to align the current camera to that object. Thus you can check whether the scene is correctly loaded.

MiangChen, 2025-03-10:
> Hi [@MiangChen](https://github.com/MiangChen), the black screen might be caused by the poor pose of the perspective camera. You may expand the stage panel on the right, select an object, and then [...]

hi, I try this, and I can see a coordinate system indicating the position of that thing, but they are not rendered properly.

https://github.com/user-attachments/assets/f6d9ec13-784e-4e0a-a7a2-2670dfa55750

HanqingWangAI, 2025-03-10:
It seems that the references are broken. Can you check the existence of the referred usd files? You can find the reference path in the `Property` panel.

MiangChen, 2025-03-10:
i check the reference path, and there is some error.
the origin reference path is `models/layout/articulated……`, so the usd and mdl files cann't be found.

[image]

and I change the true usd path to  `../../models/layout/articulated……` manually, then the cabinet can be found and rendered.

[image]

the content in the `modesl` is `../../models` and in the `Materials` is `../../Materials`:

[image]

but this files doesn't work for some reason, so the scenes are not rendered properly.
I will try to find the reason today.

HanqingWangAI, 2025-03-11:
Hi @MiangChen , thank you for the details about the issue. It is caused due to the broken symlinks of the `Materials` and `models` folder. You can fix this by rebuilding the symlinks. Note that the simlinks are not only under the individual scene folders, there are also simlinks pointing to `Materials` folder for each model instance under `models/{type}/{category}/{instance}` for proper scene loading and rendering.

HanqingWangAI, 2025-03-11:
Hi @tianshihan818 , can you provide a script for automatic symlink fixing?

MiangChen, 2025-03-11:
> Hi [@MiangChen](https://github.com/MiangChen) , thank you for the details about the issue. It is caused due to the broken symlinks of the `Materials` and `models` folder. You can fix this by [...]

> Hi [@tianshihan818](https://github.com/tianshihan818) , can you provide a script for automatic symlink fixing?

tks, looking forward to the script for repair. I can help test the effect later if necessary

tianshihan818, 2025-03-12:
@MiangChen Hello! Thanks for your feedback! You could try this script to fix/recreate the symlinks inside the scenes folder:
``` python
import os

def create_symlinks(scenes_folder):
    if not os.path.exists(scenes_folder):
        raise FileNotFoundError(f"scenes_folder {scenes_folder} not found!")
    
    models_dir = os.path.abspath(os.path.join(scenes_folder, 'models'))
    scenes_dir = os.path.abspath(os.path.join(scenes_folder, 'scenes'))
    materials_dir = os.path.abspath(os.path.join(scenes_folder, 'Materials'))

    # create "Materials" symlink in models_dir
    for root, _, files in os.walk(models_dir):
        for _ in files:
            link_path = os.path.join(root, "Materials")
            materials_dir_relpath = os.path.relpath(materials_dir, root)
            if not os.path.islink(link_path):
                os.symlink(materials_dir_relpath, link_path)
    
    # create "Materials" and "models" symlink in scenes_dir
    for root, _, files in os.walk(scenes_dir):
        for _ in files:
            materials_link_path = os.path.join(root, "Materials")
            models_link_path = os.path.join(root, "models")
            materials_dir_relpath = os.path.relpath(materials_dir, root)
            models_dir_relpath = os.path.relpath(models_dir, root)
            if not os.path.islink(materials_link_path):
                os.symlink(materials_dir_relpath, materials_link_path)
            if not os.path.islink(models_link_path):
                os.symlink(models_dir_relpath, models_link_path)

SCENES_FOLDER = 'commercial_scenes'
create_symlinks(SCENES_FOLDER)
```

m-muaz, 2025-03-12:
Hi @HanqingWangAI, I am facing somewhat similar issue. When I try to play the table scene given in demo franka scene dir via the command 
```bash
python toolkits/grscenes_scripts/preprocess.py -f /root/Documents/grutopia_assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd -i
```
Then streaming client does not display anything and the logs I see on terminal are the following:

[image]

I also tried to load the usd file through this method (https://github.com/OpenRobotLab/GRUtopia/issues/41#issuecomment-2708892760) but then I get this error:

[image]

Is there anything I am doing wrong here?@MiangChen can you guide me as how you are able to play the scene? Any help will be appreciated!

(P.S: I downloaded the assets at a different directory but I ran the `python grutopia/set_assets.py` to set the assets location and I also added the link to scene materials in the bashrc file (pic attached below)

[image]

MiangChen, 2025-03-13:
hi, I try your command`python toolkits/grscenes_scripts/preprocess.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd` in my terminal, and the error is `No preprocess method specified!`

I try the another command `python toolkits/grscenes_scripts/play_scene.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd`. 
Although there would be some red error saying "No space left on device", don't worry, just wait, and then you can see the demo scene.

[image]

[image]

MiangChen, 2025-03-13:
> [@MiangChen](https://github.com/MiangChen) Hello! Thanks for your feedback! You could try this script to fix/recreate the symlinks inside the scenes folder:
> 
> import os
> 
> 
> def create_symlinks(scenes_folder):
>     if not os.path.exists(scenes_folder):
>         raise FileNotFoundError(f"scenes_folder {scenes_folder} not found!")
>     
>     models_dir = os.path.abspath(os.path.join(scenes_folder, 'models'))
>     scenes_dir = os.path.abspath(os.path.join(scenes_folder, 'scenes'))
>     materials_dir = os.path.abspath(os.path.join(scenes_folder, 'Materials'))
> 
>     # create "Materials" symlink in models_dir
>     for root, _, files in os.walk(models_dir):
>         for _ in files:
>             link_path = os.path.join(root, "Materials")
>             materials_dir_relpath = os.path.relpath(materials_dir, root)
>             if not os.path.islink(link_path):
>                 os.symlink(materials_dir_relpath, link_path)
>     
>     # create "Materials" and "models" symlink in scenes_dir
>     for root, _, files in os.walk(scenes_dir):
>         for _ in files:
>             materials_link_path = os.path.join(root, "Materials")
>             models_link_path = os.path.join(root, "models")
>             materials_dir_relpath = os.path.relpath(materials_dir, root)
>             models_dir_relpath = os.path.relpath(models_dir, root)
>             if not os.path.islink(materials_link_path):
>                 os.symlink(materials_dir_relpath, materials_link_path)
>             if not os.path.islink(models_link_path):
>                 os.symlink(models_dir_relpath, models_link_path)
> 
> 
> SCENES_FOLDER = 'commercial_scenes'
> create_symlinks(SCENES_FOLDER)

Hi, @tianshihan818 @HanqingWangAI , I have improve the code to deal with several cases.
case1: the "Materials" and "models" are files, they should be removed first. 

[image]

case2: "Materials" amd “models" are true folders, nothing to do

```
import os

def create_symlinks(scenes_folder):
    if not os.path.exists(scenes_folder):
        raise FileNotFoundError(f"scenes_folder {scenes_folder} not found!")

    models_dir = os.path.abspath(os.path.join(scenes_folder, 'models'))
    scenes_dir = os.path.abspath(os.path.join(scenes_folder, 'scenes'))
    materials_dir = os.path.abspath(os.path.join(scenes_folder, 'Materials'))
	
    def add_link_according_to_different_cases(link_type:str, link_path, dir_relpath):
        if not os.path.islink(link_path):
            if os.path.isfile(link_path):  # in case of when “Materials" or "model" is a file
                os.remove(link_path)
                os.symlink(dir_relpath, link_path)
            elif os.path.isdir(link_path): # in case of "Materials" or "model" is the true Material/model folder
                pass
            else:  # in case of some files are missing "Materials"/"model" file, or other irrelevant folder . You need to check manually according to the log. 
                print(f"No **{link_type}** file or link in {root}, please check manually")

    
    # create "Materials" symlink in models_dir
    for root, _, files in os.walk(models_dir):
        for _ in files:
            materials_link_path = os.path.join(root, "Materials")
            materials_dir_relpath = os.path.relpath(materials_dir, root)
            add_link_according_to_different_cases(link_type="Materials", link_path=materials_link_path, dir_relpath=materials_dir_relpath)

    # create "Materials" and "models" symlink in scenes_dir
    for root, _, files in os.walk(scenes_dir):
        for _ in files:
            materials_link_path = os.path.join(root, "Materials")
            materials_dir_relpath = os.path.relpath(materials_dir, root)
            add_link_according_to_different_cases(link_type="Materials", link_path=materials_link_path, dir_relpath=materials_dir_relpath)

            models_link_path = os.path.join(root, "models")
            models_dir_relpath = os.path.relpath(models_dir, root)
            add_link_according_to_different_cases(link_type="models", link_path=models_link_path, dir_relpath=models_dir_relpath)
Q

if __name__ == '__main__':
    SCENES_FOLDER = '/home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes'
    create_symlinks(SCENES_FOLDER)
```

finally the scene can be rendered properly:

[image]

MiangChen, 2025-03-13:
but ther are still some models can't be rendered properly, so they are red.
```
[Error] [omni.hydra] Failed to create MDL shade node for prim '/Root/Meshes/Animation/electriccooker/model_7e3a193ebd67608819a5b8a554f4301e_0/Looks/MI_657734774a62b000011d0b75/Shader'. createMdlModule failed.
 [Warning] [rtx.neuraylib.plugin] Loading MdlModule to DB (OmniUsdMdl) failed: ::Z73file_3A::home::ubuntu::PycharmProjects::GRUtopia::grutopia::assets::scenes::Z60GRScenes_2D100::home_scenes::scenes::MV7J6NIKTKJZ2AABAAAAADQ8_usd::models::object::articulated::teatable::Z1D_364cdcb0b12dc0bca00ef5ce133da283b::Materials::MI_6576cd09a9943b0001ca97b1
2025-03-13 08:55:12 [111,213ms] [Warning] [rtx.neuraylib.plugin] Loading transaction committed (this thread). MdlModule is NOT in the DB (OmniUsdMdl): ::Z73file_3A::home::ubuntu::PycharmProjects::GRUtopia::grutopia::assets::scenes::Z60GRScenes_2D100::home_scenes::scenes::MV7J6NIKTKJZ2AABAAAAADQ8_usd::models::object::articulated::teatable::Z1D_364cdcb0b12dc0bca00ef5ce133da283b::Materials::MI_6576cd09a9943b0001ca97b1
[Error] [omni.usd] USD_MDL: in LoadModule at line 247 of ../../source/plugins/usdMdl/neuray.cpp -- 'rtx::neuraylib::MdlModuleId' for '/home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADQ8_usd/models/object/articulated/teatable/64cdcb0b12dc0bca00ef5ce133da283b/Materials/MI_6576cd09a9943b0001ca97b1.mdl' is Invalid

[Error] [omni.usd] USD_MDL: in GetSdrFromDiscoveryResult at line 178 of ../../source/plugins/usdMdl/moduleRegistry.cpp -- Module: '/home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes/scenes/MV7J6NIKTKJZ2AABAAAAADQ8_usd/models/object/articulated/teatable/64cdcb0b12dc0bca00ef5ce133da283b/Materials/MI_6576cd09a9943b0001ca97b1.mdl' with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.
```
These should be mainly caused by `XXX.mdl with version '1' not found in 'MdlModuleRegistry::ModuleDataMap'.` 、`'rtx::neuraylib::MdlModuleId' for XXX.mdl is Invalid` and `Failed to create MDL shade node for prim XXX createMdlModule failed.`

m-muaz, 2025-03-13:
> hi, [@m-muaz](https://github.com/m-muaz) I try your command in my terminal, and the error is `No preprocess method specified!`
> 
> ```
> python toolkits/grscenes_scripts/preprocess.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd` 
> ```
> 
> [image]
> 
> I try to add a parameter `-i` , and there is still error:
> 
> [image]
> 
> So I try another command
> 
> ```
> python toolkits/grscenes_scripts/play_scene.py -f /home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/demo_scenes/franka_mocap_teleop/table_scene.usd
> ```
> 
> Although there may be some red error saying "No space left on device", don't worry, just wait, and then you can see the demo scene.
> 
> [image]
> 
> [image]

Thanks for trying @MiangChen . I got the same error when I tried to preprocess the table scene with `-i` flag and got the `invalid prim access` error. I will try the play_scene cmd again with the table and hopefully as you also told it will take time to work. One more q @MiangChen, what GPU are you using to run the simulation?

MiangChen, 2025-03-14:
@m-muaz hi, I installed the Ubuntu 22.04 Desktop on my workstation, with 4070 ti super 16Gb. The driver version is 535.216.01

m-muaz, 2025-03-14:
hi @HanqingWangAI , I have a question did you guys tried playing the scenes in nvidia issac container version? I am still having trouble loading scene using the `play_scene.py` script, it seem like it has some issue working with docker. I'll try to debug the problem tomorrow but, just want to check if anything like this has been encountered before. Thanks!

superjuices, 2025-03-14:
After I processed the above summary, I encountered the following error when loading home_scenes：

[image]

superjuices, 2025-03-14:
> After I processed the above summary, I encountered the following error when loading home_scenes：
> 
> [image]

Does anyone of you know how to solve it? Thanks!

MiangChen, 2025-03-16:
> After I processed the above summary, I encountered the following error when loading home_scenes：
> 
> [image]

hi, I think this is a loading path error.
You can try this conversation above to fix these error.

> > [@MiangChen](https://github.com/MiangChen) Hello! Thanks for your feedback! You could try this script to fix/recreate the symlinks inside the scenes folder:
> > ```
> > import os
> > 
> > 
> > def create_symlinks(scenes_folder):
> >     if not os.path.exists(scenes_folder):
> >         raise FileNotFoundError(f"scenes_folder {scenes_folder} not found!")
> >     
> >     models_dir = os.path.abspath(os.path.join(scenes_folder, 'models'))
> >     scenes_dir = os.path.abspath(os.path.join(scenes_folder, 'scenes'))
> >     materials_dir = os.path.abspath(os.path.join(scenes_folder, 'Materials'))
> > 
> >     # create "Materials" symlink in models_dir
> >     for root, _, files in os.walk(models_dir):
> >         for _ in files:
> >             link_path = os.path.join(root, "Materials")
> >             materials_dir_relpath = os.path.relpath(materials_dir, root)
> >             if not os.path.islink(link_path):
> >                 os.symlink(materials_dir_relpath, link_path)
> >     
> >     # create "Materials" and "models" symlink in scenes_dir
> >     for root, _, files in os.walk(scenes_dir):
> >         for _ in files:
> >             materials_link_path = os.path.join(root, "Materials")
> >             models_link_path = os.path.join(root, "models")
> >             materials_dir_relpath = os.path.relpath(materials_dir, root)
> >             models_dir_relpath = os.path.relpath(models_dir, root)
> >             if not os.path.islink(materials_link_path):
> >                 os.symlink(materials_dir_relpath, materials_link_path)
> >             if not os.path.islink(models_link_path):
> >                 os.symlink(models_dir_relpath, models_link_path)
> > 
> > SCENES_FOLDER = 'commercial_scenes'
> > create_symlinks(SCENES_FOLDER)
> > ```
> 
> Hi, [@tianshihan818](https://github.com/tianshihan818) [@HanqingWangAI](https://github.com/HanqingWangAI) , I have improve the code to deal with several cases. case1: the "Materials" and "models" are files, they should be removed first.
> 
> [image]
> 
> case2: "Materials" amd “models" are true folders, nothing to do
> 
> Then the code is:
> 
> ```
> import os
> 
> 
> def create_symlinks(scenes_folder):
>     if not os.path.exists(scenes_folder):
>         raise FileNotFoundError(f"scenes_folder {scenes_folder} not found!")
> 
>     models_dir = os.path.abspath(os.path.join(scenes_folder, 'models'))
>     scenes_dir = os.path.abspath(os.path.join(scenes_folder, 'scenes'))
>     materials_dir = os.path.abspath(os.path.join(scenes_folder, 'Materials'))
> 	
>     def add_link_according_to_different_cases(link_type:str, link_path, dir_relpath):
>         if not os.path.islink(link_path):
>             if os.path.isfile(link_path):  # in case of when “Materials" or "model" is a file
>                 os.remove(link_path)
>                 os.symlink(dir_relpath, link_path)
>             elif os.path.isdir(link_path): # in case of "Materials" or "model" is the true Material/model folder
>                 pass
>             else:  # in case of some files are missing "Materials"/"model" file, or other irrelevant folder . You need to check manually according to the log. 
>                 print(f"No **{link_type}** file or link in {root}, please check manually")
> 
>     
>     # create "Materials" symlink in models_dir
>     for root, _, files in os.walk(models_dir):
>         for _ in files:
>             materials_link_path = os.path.join(root, "Materials")
>             materials_dir_relpath = os.path.relpath(materials_dir, root)
>             add_link_according_to_different_cases(link_type="Materials", link_path=materials_link_path, dir_relpath=materials_dir_relpath)
> 
> 
>     # create "Materials" and "models" symlink in scenes_dir
>     for root, _, files in os.walk(scenes_dir):
>         for _ in files:
>             materials_link_path = os.path.join(root, "Materials")
>             materials_dir_relpath = os.path.relpath(materials_dir, root)
>             add_link_according_to_different_cases(link_type="Materials", link_path=materials_link_path, dir_relpath=materials_dir_relpath)
> 
>             models_link_path = os.path.join(root, "models")
>             models_dir_relpath = os.path.relpath(models_dir, root)
>             add_link_according_to_different_cases(link_type="models", link_path=models_link_path, dir_relpath=models_dir_relpath)
> 
> 
> 
> if __name__ == '__main__':
>     SCENES_FOLDER = '/home/ubuntu/PycharmProjects/GRUtopia/grutopia/assets/scenes/GRScenes-100/home_scenes'
>     create_symlinks(SCENES_FOLDER)
> ```
> 
> finally the scene can be rendered properly:
> 
> [image]

tianshihan818, 2025-03-17:
> hi [@HanqingWangAI](https://github.com/HanqingWangAI) , I have a question did you guys tried playing the scenes in nvidia issac container version? I am still having trouble loading scene using the [...]

Hi @m-muaz ! We have tried playing the scenes in nvidia isaacsim docker, but there are some unknown issues leading to failure. I think the main reason is, we use relative paths when referencing the models, materials and textures inside the whole scene usd file. It's by design for convenience to transfer the scene usd files to anyone with different hosts. This works fine with the isaacsim standalone version, but fails with the isaacsim container. I have some guesses that might fix the problem:

1. Replace the relative paths with absolute paths when referencing models and materials
or
3. Remount the directory paths between the container and host, then rebuild the soft symlinks

I will do some tests tonight to verify it.

superjuices, 2025-03-18:
> > After I processed the above summary, I encountered the following error when loading home_scenes：
> > [image]
> 
> Does anyone of you know how to solve it? Thanks!

t

> > After I processed the above summary, I encountered the following error when loading home_scenes：
> > [image]
> 
> hi, I think this is a loading path error. You can refer to the previous record of this conversation to fix the bug: [#41 (comment)](https://github.com/OpenRobotLab/GRUtopia/issues/41#issuecomment-2720423314)

Now I can get the scene to display properly, thanks !

m-muaz, 2025-03-19:
> > hi [@HanqingWangAI](https://github.com/HanqingWangAI) , I have a question did you guys tried playing the scenes in nvidia issac container version? I am still having trouble loading scene using the `play_scene.py` script, it seem like it has some issue working with docker. I'll try to debug the problem tomorrow but, just want to check if anything like this has been encountered before. Thanks!
> 
> Hi [@m-muaz](https://github.com/m-muaz) ! We have tried playing the scenes in nvidia isaacsim docker, but there are some unknown issues leading to failure. I think the main reason is, we use relative paths when referencing the models, materials and textures inside the whole scene usd file. It's by design for convenience to transfer the scene usd files to anyone with different hosts. This works fine with the isaacsim standalone version, but fails with the isaacsim container. I have some guesses that might fix the problem:
> 
> 1. Replace the relative paths with absolute paths when referencing models and materials
>    or
> 2. Remount the directory paths between the container and host, then rebuild the soft symlinks
> 
> I will do some tests tonight to verify it.

Thanks @tianshihan818  for your support and suggestions. I'll try your suggestions later today. Also, in case you already verified it then, let me know.

m-muaz, 2025-03-23:
> Hi [@m-muaz](https://github.com/m-muaz) ! We have tried playing the scenes in nvidia isaacsim docker, but there are some unknown issues leading to failure. I think the main reason is, we use [...]

Hi @tianshihan818 checking in if you were able to narrow down the problem for loading scenes in docker container.