InternRobotics/InternUtopia #44: some tips about NVIDIA Driver version and installation

MiangChen, 2025-03-12:
I change my system from 20.04 into Ubuntu 22.04(kernel 6.8.0) these day.

I find that the recommended NVIDIA driver version `535.129.03` mentioned in README.md doesn't work. 
And the recommand driver version for Ubuntu 22.04 is `535.216.01`, see [ref1](https://forums.developer.nvidia.com/t/nvidia-driver-nvidia-linux-x86-64-535-129-03-run-installation-failed-on-ubuntu-22-04/304564/3?u=2594416505), [ref2](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/requirements.html). 

Besides, it's necessary to choose *install opengl* when install the NVIDIA driver, althougt there are many blogs saying opengl is useless... 

if `opengl` is not installed, isaacsim will encounter `fail to create any GPU devices` and `No device could be created` errors: 
[image]
[image]

HanqingWangAI, 2025-03-12:
Hi @MiangChen , thank you very much for the valuable feedback! We will double check the compatibility and further refine the docs. Your support for GRUtopia means a lot to us, and we truly appreciate it! 😊

uzuku, 2025-03-18:
The corresponding doc has been updated. Thank you for your feedback!

closed 2025-03-18 by uzuku