InternRobotics/InternUtopia #47: When the robot contains sensors, the CPU is fully loaded

hufalong, 2025-03-28:
grutopia/demo/mobile_manipulation.py or grutopia/demo/h1_locomotion.py When running, as long as sensors are added to the robot, the CPU will run full and the media terminal will be too laggy.

warning will appear midway, I don't know if it will have any impact
`failed to cook GPU-compatible mesh, collision detection will fall back to cpu`

Or if there is any direction for investigation, please provide me with some troubleshooting ideas

configuration
GPU: RXT 4090
CPU: I9-14900k 
Memory: 125G
GRUtopia: 2.0.0

uzuku, 2025-03-31:
Hi @369-hu This is a known issue about the camera, that the point cloud generation from depth camera data is inefficient and cause a 100% cpu usage.
We are working on this problem.
As a workaround, if the point cloud data is not necessary in your case, you can turn it off by commenting out the following lines:

https://github.com/OpenRobotLab/GRUtopia/blob/31ce4a2558ee9d4c3001bc9bd83575e0b0cb5dc9/grutopia_extension/sensors/rep_camera.py#L101-L105

uzuku, 2025-05-12:
Hi @369-hu We've resolved the problem in v2.1.0. You can try it out.

closed 2025-05-12 by uzuku