InternRobotics/InternUtopia #48: Sensor rendering issues

001SCH, 2025-03-31:
Hi！ The rendering of images and depth in your scene is abnormal, but the scene that comes with NVIDIA does not have this problem.Can you help check it?
[image]

superjuices, 2025-04-01:
The scene path is：GRScenes-100/commercial_scenes/scenes/MV4AFHQKTKJZ2AABAAAAADQ8_usd/start_result_navigation.usd，The way to obtain the image is through ROS messages，using Nova_Carter_ROS that comes with isaacsim:
[image]

The scene from NVIDIA is hospital, like this:

[image]

isaacsim version is 4.2

superjuices, 2025-04-01:
> The scene path is：GRScenes-100/commercial_scenes/scenes/MV4AFHQKTKJZ2AABAAAAADQ8_usd/start_result_navigation.usd，The way to obtain the image is through ROS messages，using Nova_Carter_ROS that comes [...]

The specific scene should not be MV4AFHQKTKJZ2AABAAAAADQ8_usd, but you can select some scenes in GRScenes-100 to check

uzuku, 2025-04-01:
Hi @superjuices unfortunately we are not very familiar with the ROS messages and its usage. Could you please provide a minimum set of steps/codes to get some image frames with ROS messages so we can quickly reproduce your problem.

superjuices, 2025-04-02:
> Hi [@superjuices](https://github.com/superjuices) Could you please provide a minimum set of steps/codes to get some image frames with ROS messages so we can quickly reproduce your problem.

Hi, this is very simple, just follow two steps:
1. Configure ROS2 according to this page: https://docs.isaacsim.omniverse.nvidia.com/4.2.0/installation/install_ros.html#isaac-sim-app-install-ros
2. Open isaacsim, load the environment assets, and put the built-in Nova_Carter_ROS.usd (as shown in the picture commented above) and the environment together, and click play
[image]
4. Enter ros2 topic list in the terminal to see if there is a corresponding camera node, then open rivz2 and add the corresponding image display
[image]
[image]

Hope you can find a solution！

uzuku, 2025-04-16:
@superjuices Sorry for the late response. Thank you for your kind reply, and we are trying to reproduce the case internally.

superjuices, 2025-04-16:
> [@superjuices](https://github.com/superjuices) Sorry for the late response. Thank you for your kind reply, and we are trying to reproduce the case internally.

Thank you for your reply, I still look forward to your solution to this problem!