InternRobotics/InternUtopia #53: [Bug]: Bug when try to start a  scence with controllable robot

wuzeyuuu, 2025-04-23:
### Bug Description

when I try to change the scene_asset_path parameter in demo/g1_locomotion.py to get a controllable robot in home/commercial scences. I met this error.My env is Ubuntu 22.04.5 LTS, Isaac Sim 4.2

[image]

I have tried 2 solution mentioned at https://github.com/OpenRobotLab/GRUtopia/issues/30#issuecomment-2690316295, Neither of them works. Isacc sim 4.1 met the same error to 4.2.  "from pxr import Usd" met a python extension error

[image]

[image]

### Steps to Reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### Expected Behavior

normally start

### Environment

- OS: [e.g. Ubuntu 22.04]
- GPU/CPU [e.g. A100, RTX 4090, i9-14900K]
- GPU-driver version

### Release version or Commit ID

ubuntu 22.04
isaac sim 4.2

wuzeyuuu, 2025-04-23:
https://github.com/OpenRobotLab/GRUtopia/issues/41#issuecomment-2720423314
I have also tried this method, still not work. I am sure that the path has the file exist.

uzuku, 2025-04-23:
Hi @wuzeyuuu Could you provide the modified `g1_locomotion.py` file that can reproduce this error?

closed 2025-04-27 by uzuku