InternRobotics/InternUtopia #58: [Question]: Teleoperating with VisionPro 指的是使用苹果的 VisionPro设备吗

ting-zheng, 2025-04-29:
### Question

 您好，请问Teleoperating with VisionPro这个需要具体的实体设备吗

allenpjlab, 2025-04-29:
@ting-zheng Yes, it requires Apple Vision Pro hardware to operate. Please ensure you have access to the device before running the demo.

ting-zheng, 2025-04-29:
> 是的，它需要 Apple Vision Pro 硬件才能运行。在运行演示之前，请确保您可以访问该设备。

那请问Teleoperating with Mocap这个，在其他机器人上适用吗，比如：宇树的G1

ting-zheng, 2025-05-01:
> > 那请问Teleoperating with Mocap这个，在其他机器人上适用吗，比如：宇树的G1
> 
> 对你的目标感到好奇。您打算捕捉完整的 G1 身体，还是只捕捉手？
我是想捕捉手部的动作，采集一些抓取数据之类的

JiantongChen, 2025-05-06:
> > > 那请问Teleoperating with Mocap这个，在其他机器人上适用吗，比如：宇树的G1
> > 
> > 
> > 对你的目标感到好奇。您打算捕捉完整的 G1 身体，还是只捕捉手？
> > 我是想捕捉手部的动作，采集一些抓取数据之类的

The HAMER algorithm currently enables the extraction of keypoint-level data for hand joints. Technically, it is feasible to achieve consistent hand pose mapping between humans and the GR1 robot. However, since HAMER solely provides hand-specific data and lacks critical limb-state information (e.g., arm joint angles or torso posture), alternative methodologies must be devised to drive the GR1’s end effector to its target configuration.

closed 2025-05-06 by ting-zheng

allenpjlab, 2025-06-09:
> Is it possible to use pico4 instead of vision pro

No it likely won't work since SDKs for operating VisionPro and Pico4 are distinct, meaning their usage workflows differ accordingly.

Please open an new issue if you have futher questions since this one is arleady closed.