InternRobotics/InternUtopia #61: [Question]:

jalenzz, 2025-05-16:
### Question

Hi,
May I ask which USD file is shown in the image?

[image]

Also, in what scenes are hinged objects included, and how many hinged objects are there in total?

Thanks a lot!

tianshihan818, 2025-05-19:
Hi! This scene is under the folder `commercial_scenes`, the scene ID may be `MV4AFHQKTKJZ2AABAAAAADQ8_usd`. The articulated(hinged) objects are normally existing in every scene. If you are only focused on the articulated objects, you could find them in this folder path `home_scenes/models/articulated` or `commercial_scenes/models/articulated`, including around five thousand items in total.

jalenzz, 2025-05-22:
Thanks for the reply.

closed 2025-05-22 by tianshihan818