InternRobotics/InternUtopia #62: [Question]: How effective is the interaction system in this simulator?

doraemonaaaa, 2025-05-19:
### Question

Question1: Are the characters in this simulator movable? Also, how can we determine whether the robot is able to initiate or engage in a conversation with a character?

Question2:I’ve been using Isaac Sim for about six months, but I find it quite difficult to use. I’m planning to conduct research in Human-Computer Interaction (HCI), and since we want more flexibility and scalability for interaction design, and we'll build some robot simulation,  we’re considering switching to Unity. What do you think of this approach?

I’ve built a demo in Unity and found that creating interactive scenes is not only quick but also far more adaptable compared to other simulators.

doraemonaaaa, 2025-05-19:
I'm really looking forward to your answer

doraemonaaaa, 2025-05-23:
I recently finished building a traffic system and a weather system in Unity, and I’m really impressed by how powerful Unity is for simulating real-world environments. It seems like a great tool for Human-Computer Interaction (HCI) simulations.

Since I’m still new to this field, I’m curious—why isn’t Unity more commonly used as a base engine for realistic simulations? I’d love to hear your thoughts or experiences if you’ve worked on something similar!

HanqingWangAI, 2025-05-28:
Hi @doraemonaaaa , thank you for reaching out! I’d like to share my perspective—this is just my personal take, so feel free to take it with a grain of salt. :)

Generally, the choice of engine depends on your specific goals. Unity is excellent for developing single-threaded applications like HCI tools or interactive prototypes. However, for robotics learning, it might not be the best fit due to two key limitations:

- ​​Performance Constraints:​​ Unity can be prohibitively slow for real-time tasks like online training or large-scale data collection, which are often critical in robotics workflows.
- ​​Physics Accuracy:​​ While Unity offers physics engines (e.g., PhysX), they’re optimized for gameplay rather than precision. Robotics simulations typically require highly accurate physics models, which may not be readily achievable in Unity.

That said, Unity’s strengths in visualization and rapid prototyping could still complement certain aspects of robotics.

closed 2025-06-13 by uzuku