InternRobotics/InternUtopia #63: [Enhancement]: Isaac Sim Rendering Issue on RTX 50 series

996-PYG, 2025-05-22:
### What feature or enhancement are you proposing?

[image]
GRUtopia fails to render properly on my RTX 5090D GPU. This appears to be an issue with Isaac Sim itself. We hope future updates will enable GRUtopia to support the 50 Series GPUs."
https://forums.developer.nvidia.com/t/isaac-sim-rendering-issue-on-rtx-50-series/329300
```
from grutopia.core.config import Config, SimConfig
from grutopia.core.gym_env import Env
from grutopia.core.runtime import SimulatorRuntime
from grutopia.core.util import has_display
from grutopia.macros import gm
from grutopia_extension import import_extensions
from pprint import pprint
from PIL import Image
from grutopia_extension.configs.robots.h1 import (
    H1RobotCfg,
    h1_camera_cfg,
    h1_tp_camera_cfg,
    move_along_path_cfg,
    move_by_speed_cfg,
    rotate_cfg,
)
from grutopia_extension.configs.tasks import (
    SingleInferenceEpisodeCfg,
    SingleInferenceTaskCfg,
)

headless = False
if not has_display():
    headless = True

h1_1 = H1RobotCfg(
    position=(0.0, 0.0, 1.05),
    controllers=[
        move_by_speed_cfg,
        move_along_path_cfg,
        rotate_cfg,
    ],
    sensors=[
        h1_camera_cfg.update(name='camera', resolution=(320, 240), enable=True),
        h1_tp_camera_cfg.update(enable=True),
    ],
)

config = Config(
    simulator=SimConfig(physics_dt=1 / 240, rendering_dt=1 / 240, use_fabric=False),
    task_config=SingleInferenceTaskCfg(
        episodes=[
            SingleInferenceEpisodeCfg(
                scene_asset_path=gm.ASSET_PATH + '/scenes/empty.usd',
                scene_scale=(0.01, 0.01, 0.01),
                robots=[h1_1],
            ),
        ],
    ),
)

print(config.model_dump_json(indent=4))

sim_runtime = SimulatorRuntime(config_class=config, headless=headless, native=headless)

import_extensions()
# import custom extensions here.

env = Env(sim_runtime)
obs, _ = env.reset()
print(f'========INIT OBS{obs}=============')

path = [(1.0, 0.0, 0.0), (1.0, 1.0, 0.0), (3.0, 4.0, 0.0)]
i = 0

move_action = {move_along_path_cfg.name: [path]}

while env.simulation_app.is_running():
    i += 1
    action = move_action
    obs, _, terminated, _, _ = env.step(action=action)
    if i % 100 == 0:
        print(i)
        if i == 100:
            print("obs:")
            # pprint(obs)
            pprint(obs["sensors"]["camera"]["rgba"])
            image = Image.fromarray(obs["sensors"]["camera"]["rgba"],'RGBA')
            image.show()
env.close()
```

### Motivation

uzuku, 2025-05-22:
Hi @996-PYG Thank you for your test and information. We'll track the new release of Isaac Sim once available.

uzuku, 2025-08-07:
Hi @ruofei0929 Thank you for your feedback. Isaac Sim 5.0 is now in pre-release stage, and we're testing against it to evaluate the workload of adaptation. We expect to release support for Isaac Sim 5.0 once our adaptation is completed upon its stable version.

youthHan, 2025-12-12:
Looking forward to the solution.