InternRobotics/InternUtopia #65: [Question]: 为什么isaacsim加载场景scenes中的usd文件无法显示场景，只有一个球形，如下图

RongkunYang, 2025-06-08:
### Question

[image]

RongkunYang, 2025-06-09:
[3,383,535ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/Root/Looks/DayMaterial/DayMaterial'. Empty identifier: '' 	and/or subIdentifier: ''

最后terminal有上面这条报错信息

tianshihan818, 2025-06-09:
> [3,383,535ms] [Error] [omni.hydra] Failed to create MDL shade node for prim '/Root/Looks/DayMaterial/DayMaterial'. Empty identifier: '' and/or subIdentifier: ''
> 
> 最后terminal有上面这条报错信息

你好！这个现象很有可能是这个场景 USD 中所引用的材质、模型路径未被正确识别并加载。可以先参考这个 [issue#41](https://github.com/OpenRobotLab/GRUtopia/issues/41) 里的讨论进行检查确认，并使用这个脚本对场景 USD 中的 reference 路径相关问题进行修复。

RongkunYang, 2025-06-09:
好的，谢谢您的及时回复，我先检查一下。想请问一下这个与isaacsim的版本没有关系对吧，本来想重新安装isaacsim4.2试一下，不过现在isaacsim4.2好像无法直接用omniverse launcher安装了

tianshihan818, 2025-06-09:
> 好的，谢谢您的及时回复，我先检查一下。想请问一下这个与isaacsim的版本没有关系对吧，本来想重新安装isaacsim4.2试一下，不过现在isaacsim4.2好像无法直接用omniverse launcher安装了

@RongkunYang 这个和版本应该无关，我看您使用的是 4.5 版本，我前几天刚在这个版本上测试加载场景是可以的。

RongkunYang, 2025-06-09:
您好，我测试了您给的脚本，不过加载场景的过程中还是会遇到如下的报错信息：
 Failed to create MDL shade node for prim '/Root/Meshes/Animation/trashcan/model_a460cee7aa4a89d353cc65a7104381d7_0/Looks/MI_6576cd05ade22a0001b57384/Shader'. createMdlModule failed.

请问这个是什么原因呢？

RongkunYang, 2025-06-09:
[image]

RongkunYang, 2025-06-09:
加载好之后场景中有一部分物体没有材质，只能显示红色的，请问您有解决办法吗

RongkunYang, 2025-06-09:
非常感谢，这样就能够正常加载场景了

RongkunYang, 2025-06-09:
请问如果需要将blender中的资产转化为USD格式，grutopia有没有提供相关的转换脚本呢

tianshihan818, 2025-06-10:
> 请问如果需要将blender中的资产转化为USD格式，grutopia有没有提供相关的转换脚本呢

@RongkunYang 你好，GRUtopia 目前没有相关转换脚本