InternRobotics/InternUtopia #67: [Question]: 关于visionPro遥操作的疑问

RongkunYang, 2025-06-18:
### Question

在实现了homebrew以及一系列certificate的安装之后，如何将ubuntu系统与visionPro建立连接呢，下面的步骤中在实际操作时“1. Open Settings and install the downloaded profile”， 好像visionPro中没有这个选项
[image]

uzuku, 2025-06-19:
Hi @RongkunYang In the step

```
cd $(mkcert -CAROOT) && python -m http.server
```

an http server is started in ubuntu. And if Vision Pro is in the same network with the Ubuntu host, you can use the Safari in Vision Pro to access the server by `http://{ubuntu_host_ip}:8000` (where 8000 is the default port of `python -m http.server`) and download the cert file in the web page.

After downloading the files to Vision Pro, you can try opening the cert file directly, or opening the Settings to see if there's a hint for the downloaded cert to help you install it.

RongkunYang, 2025-06-19:
另外就是visionpro在之前已经有其他主机安装过一次证书，这次下载完之后进入设置->通用->关于本机->证书设置好像没有提示可以安装本次下载的文件的选项

uzuku, 2025-06-23:
Hi @RongkunYang 

> 您好，在访问这个网页中有两个证书选项，具体应该下载哪个呢？

rootCA.pem should be downloaded and installed in VisionPro. Key file should only be kept in the server side.

> 另外就是visionpro在之前已经有其他主机安装过一次证书，这次下载完之后进入设置->通用->关于本机->证书设置好像没有提示可以安装本次下载的文件的选项

As the doc says, there're 2 steps to complete the cert setup:

> 1. Open Settings and install the downloaded profile
> 2. Settings > General > About > Certificate Trust Settings. Under “Enable full trust for root certificates”, turn on trust for the certificate.

In the first step, open the Settings and there should be a hint guiding you on installing the cert. Or you can try opening the cert file directly to see if it is associated to the installation process.

In the second step, you should be able to find and trust the cert installed in step one through Settings > General > About > Certificate Trust Settings.

closed 2025-07-11 by uzuku