InternRobotics/InternUtopia #69: 运行h1_locomotion demo出现xcb报错

ByZ0e, 2025-07-09:
请问安装好后运行h1_locomotion demo出现xcb报错是什么原因？

Starting kit application with the following args:  ['/home/aios/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py', '/home/aios/isaacsim/apps/isaacsim.exp.base.python.kit', '--/app/tokens/exe-path=/home/aios/isaacsim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--/app/installSignalHandlers=0', '--ext-folder', '/home/aios/isaacsim/exts', '--ext-folder', '/home/aios/isaacsim/apps', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/aios/isaacsim/kit/logs/Kit/Isaac-Sim Python/4.5/kit_20250709_125518.log
2025-07-09 04:55:18 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.025s] [ext: omni.kit.async_engine-0.0.1] startup
[0.221s] [ext: omni.metrics.core-0.0.1] startup
[0.221s] [ext: omni.client.lib-1.0.0] startup
[0.236s] [ext: omni.blobkey-1.1.2] startup
[0.236s] [ext: omni.stats-1.0.1] startup
[0.237s] [ext: omni.datastore-0.0.0] startup
[0.240s] [ext: omni.client-1.2.2] startup
[0.243s] [ext: omni.ujitso.default-1.0.0] startup
[0.244s] [ext: omni.hsscclient-1.1.1] startup
2025-07-09 04:55:18 [236ms] [Warning] [omni.datastore] OmniHub is inaccessible
[0.246s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.247s] [ext: omni.assets.plugins-0.0.0] startup
[0.247s] [ext: omni.gpu_foundation-0.0.0] startup
[0.251s] [ext: carb.windowing.plugins-1.0.0] startup
[0.255s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.163.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 29b3c167.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-139-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-14900K
| Cores: 24 | Logical Cores: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128570 | Free Memory: 60056
| Total Page/Swap (MB): 117187 | Free Page/Swap: 117187
|---------------------------------------------------------------------------------------------|
[0.672s] [ext: omni.kit.pipapi-0.0.0] startup
[0.673s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.673s] [ext: omni.pip.compute-1.4.5] startup
[0.673s] [ext: omni.materialx.libs-1.0.6] startup
[0.678s] [ext: omni.kit.telemetry-0.5.1] startup
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
Fatal Python error: Segmentation fault

Thread 0x00007fc8e110e700 (most recent call first):
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/selectors.py", line 469 in select
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/asyncio/base_events.py", line 1871 in _run_once
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/asyncio/base_events.py", line 603 in run_forever
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/site-packages/grutopia/core/util/async_req.py", line 36 in start_loop
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/threading.py", line 953 in run
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fc934862740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/aios/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/_itelemetry.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/aios/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/__init__.py", line 8 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/aios/isaacsim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/aios/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/aios/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/aios/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
  File "/home/aios/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/home/aios/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 427 in _start_app
  File "/home/aios/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 212 in __init__
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/site-packages/grutopia/core/runtime/__init__.py", line 51 in __init__
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/site-packages/grutopia/demo/h1_locomotion.py", line 52 in <module>
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/runpy.py", line 86 in _run_code
  File "/home/aios/miniconda3/envs/grutopia/lib/python3.10/runpy.py", line 196 in _run_module_as_main

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, yaml._yaml (total: 14)

BrigantineX, 2025-07-09:
Please provide more information for us to investigate the issue:

According to https://grutopia.github.io/usage/get_started/installation.html, there are three ways of installation:
1. Install from source
2. Install from PyPI
3. Install with Docker
which way did you use? 
did you install other package in conda after installation?
What version of iassc-sim are you using?

ByZ0e, 2025-07-09:
I installed from PyPI. No. I have tried both isaacsim-4.2.0 and isaacsim-4.5.0. Both of them don't work.
The isaac-sim.sh can be started up successfully.

ByZ0e, 2025-07-09:
Thanks a lot! It works well.

closed 2025-07-09 by ByZ0e